Oriolac/seu #39: Review Documentation

quimpm, 2021-12-10:
Final Review of the documentation.

quimpm, 2021-12-29:
- WindFarm Project
   - Project layout
   - Slides final presentation
- Supervision
   - Sequence Diagram Arduino + ChibiOS 
   - Arduino
       - Merge of both codes for Arduino Uno
   - ESP01 Protocol
   - CibiOS
      - Comentar introduccio 
      - Merge
   - Move Chibi to ChibiOS
   - DataProducer1
      - Fritzzing
      - Code Of Dataproducer 1
      - Merge with MQTT
   - DataProducer2
      - Fritzzing
      - Code
      - Merge code with MQTT
      - Delete Wiring with fritzzing in LCD screen
   - ESP 01
      - Delete Humidity sensor section
      - Move LCD to Dataproducer1
    - Mqtt
       - Broker -> Getting Started
       - Fritzzing MQTT Broker
       - Flux Diagram
       - Topics tree

sergisi, 2021-12-29:
Done:
- Project Layout
- Broker -> Getting started with uMQTTBroker

sergisi, 2021-12-29:
Doing:
- All the items in Mqtt

sergisi, 2021-12-29:
Done:
- Fritzing MQTT Broker
- Flux Diagram

Oriolac, 2021-12-29:
Done: 
* Index nav
* Supervision:
   * ChibiOS
* Move Chibi to ChibiOS

quimpm, 2021-12-29:
GJ guys, okeeeei lesgoooo

closed 2021-12-29 by quimpm